Carvera-Community/Carvera_Controller #23: Configurable Autolevel Area

SergeBakharev, 2024-11-04:
Currently when auto leveling is enabled the machine probes an area using `G32` over the area encompassing the tool paths. 

It would be nice to be able to adjust the area that is probed to be a subset of the toolpath region.

A common scenario is needing to avoid clamps on top of the stock, or cases where the toolpaths might be off the stock (eg facing).

SergeBakharev, 2024-11-04:
[image] [Auto - level changes](https://trello.com/c/iVePn1TI/7-auto-level-changes)

faecorrigan, 2025-01-17:
[image]

working through the details to make it all work. When the check box for offsets is selected the program will offset the autoleveling path accordingly, when it is unselected it will default to the boundary region.

This won't solve the problem of skipping certain probe points, but it is a start

SergeBakharev, 2025-01-17:
Looks great!

closed 2025-03-22 by SergeBakharev