Carvera-Community/Carvera_Controller #45: Soft lock on file upload to machine.

SergeBakharev, 2024-11-09:
Per https://github.com/MakeraInc/CarveraController/issues/26

Looks like it would be beneficial to stage gcode files in the temp directory before uploading to the machine to prevent this issue.

Kramax, 2026-10-12:
I am encountering this issue with a linux based host, trying to upload a local to the host file. Have to kill the controller and restart. A file does appear on the carvera, but I never trust it as I worry if the entire file actually transferred. It is intermittent. The file is not large at all (like 6kb). This is with the controller version 2.2.0.

SergeBakharev, 2026-10-12:
Is the source file on some unreliable NAS/cloud storage? I'm trying to understand the relevance of your comment.

Kramax, 2026-10-12:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!
> <Carvera-Community/Carvera_Controller/issues/45/6133691038

No it is on the local SSD of the linux device running the controller. Just
wanted to give a datapoint on this as I think there must be a bug somewhere
in the code path where the machine firmware responds to the controller. I
am using wifi to communicate with the machine so there could be some
latency from that.
-Derek

closed 2026-11-13 by SergeBakharev