Carvera-Community/Carvera_Controller #55: Tooltips

SergeBakharev, 2024-11-10:
Ideally they would work like fusion360 tooltips where they have an image and text description

SergeBakharev, 2024-11-10:
[image] [UI overhaul - Tooltips](https://trello.com/c/QMDnRm9a/24-ui-overhaul-tooltips)

SergeBakharev, 2024-11-10:
If someone helps write the text/images, I can help implement it.

faecorrigan, 2024-11-10:
the bluesky version of a tooltip looks like this:

[image]

it would need to be implemented as an image with text at specific locations for localization. I think  using a base class of UI element with a few parameters that all tooltips inherit from. That way they all follow the same standard format. The base class would be kept inside its own separate file to be included. 

For practicality's sake, I would start with a vertical box with a title at the top,  image in the middle and a multiline text field at the bottom

I have a few samples of how I did it the first time I tried that I can pull up next week and see how they interact based on some of the info from Serge's link: https://stackoverflow.com/questions/34468909/how-to-make-tooltip-using-kivy

Last time I tried this there were issues with the tooltip not hovering at the correct location compared to the mouse due to a weird window offset bug.

faecorrigan, 2024-12-13:
The basic functionality for tooltips has been implemented in the tooltiptestinggrounds branch. Right now they are only implemented for buttons, but other UI elements will be easy to add. 
Exact language for the tooltips and images will need to be generated

[image]
[image]
[image]

pzich, 2025-02-12:
What's involved in getting this integrated? The bluesky tooltip like Fusion would be nice, but even just text titles would be very helpful, similar to MakeraCAM

[image]

faecorrigan, 2025-02-12:
For the controller, it is mostly just getting all the ducks in a row and then starting to fill in the tool tips.
We don't have any access to makera cam, so we won't be able to add tooltips there unfortunately

> What's involved in getting this integrated? The bluesky tooltip like Fusion would be nice, but even just text titles would be very helpful, similar to MakeraCAM
> 
> [image]

pzich, 2025-02-12:
Oh sorry I didn't mean modify MakeraCAM, just that its tooltips are pretty simple (usually just 1-2 words) but that is already much more helpful than none at all. Before getting to text styling and graphics it'd be great to have basic tooltips like they do.

faecorrigan, 2025-02-14:
Tooltip function has been merged into the dev branch and will go out with the next release of the controller. Anyone looking to help can suggest tool tip language/imagery for a particular button

SergeBakharev, 2025-05-04:
I'm going to close this issue for now. The basic functionality has been implemented, text can now be added as appropriate.

closed 2025-05-04 by SergeBakharev