Carvera-Community/Carvera_Controller #63: Translation Tech Debt

SergeBakharev, 2024-11-12:
As we start to make UI text changes we should be updating the translations.

We need to do a few things:
1. Develop a process for updating the translation strings
2. Catch up on the strings that have already been changed.

faecorrigan, 2024-11-15:
For anyone who wants to help, the translation files are plain text editable and found in carveracontroller/locales
They use two letter short names for a given language and you just need to fill out the translation of the English in the area between the "" in a text editor like visual studio, notepad++, or notepad. Do not edit any of the other formatting in the document, including spaces and tabs, and do not open the files in word or another advanced text processing program

[image]

faecorrigan, 2024-11-15:
On the tech debt side of things, I set up a program that automatically appends any new translatable strings to the po files, see the pull request. 

Any string that follows the format  .tr_("") will be automatically added to the po files
[image]
[image]

When you want to update the translation files, run update_translations.py from the carvera controller app

This will append the po files in the locals directory with any new auto translations. Adding a new auto-translation file is as simple as adding the two letter short code to update_translations. Similarly if we add any new .py or .kv files with UI in them, we would need to add them to the update_translations file

with that done, the .po files can be edited directly in any text editor. You just need to fill in the empty string corresponding to the translation. if the string is left empty it defaults to the english version

Before shipping a build, you need to run update_translations.py one last time to bake the translation information into a format readable at runtime (.mo files)

To add a new language to the dropdown: 

add it to LANG in main.py
The two letter code needs to match the name of the po file, and the long name is what shows up in the language dropdown
[image]

SergeBakharev, 2024-12-08:
Thanks Fae. I've integrated  your tool into the build tooling and written simple readme.md addition.

closed 2024-12-08 by SergeBakharev