Carvera-Community/Carvera_Controller #115: Bug: if an expected config setting does not exist on the machine, the controller will fail to connect

faecorrigan, 2024-12-15:
if a parameter does not exist on the machine config file, but the controller tries to access it, the controller gives up when connecting. 

For instance, I have been working to add atc.num_atc_touchoff to the controller as shown. 

When connecting to the machine, the connection goes through all the steps to connect, and then resets to disconnected after it syncs time and version/loads the config file

This happens because the controller is trying to access a config setting on the machine that has not been written yet. 
This will be an issue as we add new config settings to the controller for the community branch, as they will not exist on the stock machines

[image]

SergeBakharev, 2024-12-15:
I've reviewed that section of code previously when I did the Controller config feature. I think I can resolve this issue pretty easily, give me a week

SergeBakharev, 2024-12-15:
Had a review of the code, and found that as long as the new config.json entry has a `default` key the controller will correctly handle it not being present in the machine config.txt

closed 2024-12-15 by SergeBakharev