Carvera-Community/Carvera_Controller #140: Buttons are squished in jog UI

pzich, 2025-02-13:
I'm not sure if there's something wrong with how it's setup, how I'm running it, or some issue with the macOS environment (I just followed the steps in the README) but the buttons in the center of the jog control are oddly squished/overlapping compared to Makera's built app

## Community controller (via `poetry run ...`)
[image]

## Makera's built app
[image]

SergeBakharev, 2025-02-13:
Thanks for sharing. It's an issue specific to above 1x DPI scaling for some reason.

I believe I broke it when I removed a bunch of empty `Label:` components here: https://github.com/Carvera-Community/Carvera_Controller/pull/114/files#diff-34148fa80065ff15cef42d7203501ced5115b5634fc68ee0dcdf5fbf66feecb5

Leave this with me, I think it's an easy fix.

SergeBakharev, 2025-02-13:
Nope not the empty `Label:` I'll keep digging

faecorrigan, 2025-02-14:
it might be fixed with Todd's new row addition. I tried it with 1.0 scaling and it worked, however, it was defaulting to full screen instead of the smaller windowed size

https://github.com/toddmitchell/Carvera_Controller/tree/probe-prototype

SergeBakharev, 2025-02-14:
I wonder if it's an issue resulting from us using the latest version of kivvy while Makera using an ancient out of support version.
Perhaps there was a breaking change in how the `dp` values are interpreted for the UI elements. At above 1.0 scaling, I see lots of tiny UI issue, icons not centered in buttons etc.

SergeBakharev, 2025-03-22:
I think this is due to a combination of a few things, changes in kivy behavior between 1.x and 2.x versions plus
the use of `size_hint_y: None` in the `SmallRoundButton` definition. Kivy is treating the buttons on initialization as having no y height, which

I've merged this fix: #161 Please re-open an issue if it's still doesn't look right on MacOS.

I think it would be excellent to set a minimum size on some UI elements. Maybe in a future change.

PR #161 (2025-03-22): Fixed squished center buttons
My attempt to fix #140 .

1x DPI looks correct now, but 2x looks a bit smaller than it should on my machine.

Still better than the "overinflated" look previously
closed 2025-03-22 by SergeBakharev