Carvera-Community/Carvera_Controller #185: Responsive GUI is highly wanted, at the moment it looks like it's put together in 5 seconds with no thoughts at all

L0rdS474n, 2025-05-04:
Might be something on my computer but the GUI isn't precicesly professional to say it mild.

[image]

Overriding DPI settings make it both better and worse at the same time.

[image]

It might sound harse from my side but it will be the part I will spend many hours with so if it looks sloppy it just lower my happiness, is there a settings I can change for my Windows 11?

Makera original looks nice in normal window and in full screen so this should also look as nice as that original software, right?

As I said in the firmware issue, as soon as I get more time I will try to contribute to this too and not just complain ;) But until then I depend on you guys/gals out there to help me ;)

SergeBakharev, 2025-05-04:
Hi Ulf,

Can you please bit more specific in your critics? It's unclear exactly what exact issue you are experiencing, at the moment it looks like this report was written in 5 seconds 😛.

> Responsive GUI is highly wanted
The code makes use of the Kivy UI framework. This does not have capability to do layout breakpoints, the corner stone of Responsive UI. It does however have capability to describe the UI based on DPI aware units, however some of the inherited code has hard coded pixel sizes/logic. As a result we are tracking some open issues: #175 #188 

If you are interested in porting the code over to [KivyMD ](https://kivymd.readthedocs.io/en/1.0.2/components/responsivelayout/index.html) a proper responsive layout could be achieved. Otherwise the experience should be better with the resolution of those two issues.

As part of triaging #175 what we have come to learn is that the Kivy uses the DPI of the first monitor that the Window is rendered on for some of the UI attributes, and moving the Window to a monitor causes it to redraw the window with the different DPI. The issue in #175 is that some attributes are not being updated when this event occurs, the very jarring example is icon sizes are being updated but not position.

L0rdS474n, 2025-05-05:
Hehe, bouncing back 😁
As soon as my time allows I will join you on this.
Thanks for explaining why it looks like it do.

SergeBakharev, 2025-05-05:
In all seriousness we appreciate reports of any specific UI issues that we aren't already tracking. Accurate descriptions of observed problems are half the battle in resolving them. 

Similarly we welcome code contributions, please don't feel like you need to be submitting PRs to help make the project better.

Closing this issue for now. Feel free to re-open if you have more details about what you think is a problem.

closed 2025-05-05 by SergeBakharev