Carvera-Community/Carvera_Controller #189: Connecting Carvera to hidden SSID

L0rdS474n, 2025-05-05:
Currently, the terminal does support connecting to hidden SSIDs, but it cannot process complex passwords—even though the app handles them correctly—so adding support should be straightforward.

SergeBakharev, 2025-05-07:
Ulf, I think I understand what you are suggesting but others might not. Please confirm this summary:

* Carvera machine firmware supports connecting to hidden SSIDs
* Hidden SSIDs do not appear in the Controller application
* Manually configuring the Carvera to connect to a hidden SSID is possible via the machine command `wlan <ssid> <password>`
* Carvera machine firmware correctly saves complex passwords to config file, but still fails to connect to such SSID.

Thus this is an enhancement suggestion to add support to the Controller to be able to configure the wifi details of a hidden SSID. 

No enhancements to the Controller are required for handling complex password, work needs to be done in the firmware to add support. Have you already opened an issue in the firmware repo?

L0rdS474n, 2025-05-07:
Mostly correct.
wlan connand treat  spaces as a separator so it won't set the correct password. If I create a visible SSID with the same SSID and the same password and connect to it from the controller app it will connect.
App get the password correctly but the simple shell when running wlan SSID PASSWORD does not.
You can also see the command that the controller app send to simple shell but every special character such as spaces are just shown as squares.
That's why I think the asy fix would be to have the controller app to allow the user to type in the SSID and this would be the solution. That way we don't need to change the simple shell and risk breaking something else I  the process.

SergeBakharev, 2025-10-05:
Implemented in #359

PR #359 (2025-10-05): Support connecting to hidden wifi networks
Fixes #189. I've tested this on firmware 1.0.11c but can't currently test it on 2.0.0-RC1 because of the crash described in https://github.com/Carvera-Community/Carvera_Community_Firmware/issues/146. I'll troubleshoot that separately.

This is my first time working with kivy, so I'll be grateful for any feedback you have. `DualInputPopup` is a copy of `InputPopup` with modifications for the second label and input box. The code changes are modeled after the existing wifi callback code as much as possible. Storing the most recent manually-entered SSID i [... cut]
closed 2025-10-05 by SergeBakharev