Carvera-Community/Carvera_Controller #192: General UI - Screen showing current WCS visually

denunzio11, 2025-05-10:
When you go to the "run" screen, it has a visual representation of the WCS origin as defined in the program file selected.  It would be great to have another dot on the screen depicting the current machine XY WCS origin and the "G54", etc. currently in use.

WARIO2412, 2025-06-11:
I don't really understand what you mean. The current XY Origin (eg G54) actually is the dot that you see on the run screen. Only if you change the offsets on the right it shifts to where the origin is going to be if you hit run.
That means:
If you set the XY Origin manually, let's say by probing a corner and you go to the run screen, the origin dot represents the newly set origin. Only if you change the offsets, the dot will move   to a new place.

Do you mean that we should add another dot representing the current origin?

SergeBakharev, 2025-07-02:
I think the user meant showing which wcs system was active. This was implemented here: https://github.com/Carvera-Community/Carvera_Controller/pull/267 Closing this issue. @denunzio11 please reply if we have misunderstood.

closed 2025-07-02 by SergeBakharev