Carvera-Community/Carvera_Controller #197: Tool probe feedrate too high during Z calibration 3 axis cnc probe – causes 3D probe to hang in triggered state

Joqiz, 2025-05-22:
**Firmware**: Carvera.Community.Firmware.V1.0.3c1.0.5.bin  
**Machine**: Carvera Air  
**Controller**: Community controller (tested on Android & Windows)

Hi and thank you for all the great work on this firmware and controller!

I'm using a standard 3D touch probe (normally open), as shown in this guide:  
https://www.instructables.com/Carvera-Touch-Probe-Modifications/

According to the probe's specifications and common practice, the **maximum safe probing feedrate is 50 mm/min** — especially during downward movement (Z probing).

---

### Problem:
This issue is about the **tool calibration function** — the button in the community controller UI that automatically calibrates tool length (Z) using the tool setter block (probably via `M500`/`M501`).

Even when using "slow" probing, the feedrate during tool calibration appears to be too fast for sensitive 3D probes.

---

### What happens:
- The probe LED stays **red** after calibration (triggered state)
- The probe returns to normal (green) only after slight wiggling
- This suggests internal overstress of the probe mechanism
- Long term, this could lead to wear, false triggers or calibration errors

---

### Suggested improvements:
- Allow the tool probing feedrate to be **configurable**, e.g.:
  ```gcode
  config-set sd tool_probe.feedrate 50
-Apply slow feedrate only to tool 0 (3D probe), and not to normal cutting tools

Let me know if I can help test any changes — thanks again for building this amazing tool!

J

SergeBakharev, 2025-05-22:
Hi @Joqiz ,

Thanks for sharing this report. I can confirm that the Tool Calibration button in the Controller uses `M491`.

The Calibration sequence is described in the firmware method `ATCHandler::fill_cali_scripts()` [here](https://github.com/Carvera-Community/Carvera_Community_Firmware/blob/ea1f1a6b973d079bf243adba163a7cb197ca8f0f/src/modules/tools/atc/ATCHandler.cpp#L788).

During this sequence there are two probing feed speeds used, first using `probe_fast_rate` (default 300), retracting a bit (2mm default), then probing again with `probe_slow_rate` (default 60).

You can change those speeds in the Controller in Settings under "Machine - Advanced", and setting the values in "Probe Fast Speed" and "Probe Slow Speed". Screenshot attached. Alternatively you can set these values in the MDI using `config-set sd atc.probe.fast_rate_mm_m 300.0` and `config-set sd atc.probe.slow_rate_mm_m 60.0`. Don't forget to reset the machine afterwards to apply the new config.

I personally haven't experienced what you are describing, but agree that we should probably use the manufacturer's spec here and slow down the probe feed.

One option is to have a third configurable speed which is used specifically only for probes, or like you said use the slow speed only to probes. We will discuss on the discord about the best approach.

[image]

SergeBakharev, 2025-05-23:
Glad you were able to workaround the issue on your machine. 

You can join the Community Projects Discord with this link: https://discord.gg/b8szm43Z Here I've started a thread to discuss enhancements to the firmware to address this issue. 

I myself haven't written any code for the firmware, so I differ to those with experience for direction. My understanding is that the stock probe is handled differently in the Carvera Air to the Carvera, so some complexity exists to retain compatibility between the two.

As you have opened an issue in the firmware repo for this same enhancement suggestion, and since there is no bug or functionality to be added to the Controller, I'll be closing the issue here.

I look forward to seeing you in the Discord server. 😃

closed 2025-05-23 by SergeBakharev