Carvera-Community/Carvera_Controller #253: Failure to reconnect after Android tablet sleep.

jterblanche, 2025-06-24:
When Android tablet goes to sleep / auto screen lock while Carvera Community Controller is the active app, the app cannot reconnect to the Carvera Air once the tablet is woken up again. 

Requires closing / stopping the app completely and starting again in order to reconnect to Carvera Air.

jterblanche, 2025-06-28:
Thanks Serge. There is no error displayed. The tablet attempts to reconnect, displays a message saying "Loading Config", and then simply disconnects again - all the controls go dim and the state changes to disconnected.

SergeBakharev, 2025-06-28:
I think we fixed this bug in the last few days. If you don't mind can you try the latest Dev build from here: https://github.com/Carvera-Community/Carvera_Controller/releases/tag/dev

jterblanche, 2025-06-30:
Thanks! Will do.

closed 2025-07-02 by SergeBakharev