Carvera-Community/Carvera_Controller #269: Carvera Air Probe Fail on M464

jterblanche, 2025-06-30:
When using the 3D probing commands with Carvera air, the controller asks confirmation to issue a M464 ... command.

When I proceed, the Carvera Air immediately locks out with "Probe Fail".

I have tested all the variations of G38.2 and G10 L20 commands as per the link below, and they all work perfectly on X, Y as well as Z axis.
[Instructables - Carvera Touch Probe Modifications (by The Props Monster)](https://www.instructables.com/Carvera-Touch-Probe-Modifications/)

Is there a way the controller can be configured to issue G38.2 + G10 L20 (optionally with configurable P parameter, i.e. P0, P1 etc), instead of M464?

SergeBakharev, 2025-07-02:
Hmmmm I don't get this issue on my C1 on the same dev build. Are you using a Carvera Air?

SergeBakharev, 2025-07-21:
Yes, this is known issue that has been resolved in the latest Dev builds. In those builds the option to set/change to 3D Probe is present in the UI, and you don't need to use the Custom tool option.

SergeBakharev, 2025-07-30:
The latest release ([0.10.0](https://github.com/Carvera-Community/Carvera_Controller/releases/tag/v0.10.0)) should fix these issues. If it doesn't please reopen this issue.