Carvera-Community/Carvera_Controller #280: WBH04 Pendant support on Android

SergeBakharev, 2025-07-11:
I've got 80% of the way to supporting the Pendant on Android devices with USB Host capability here: https://github.com/SergeBakharev/CarveraController/tree/android_pendant_support

Unfortunately blocked by https://github.com/libusb/hidapi/issues/648

rtadams89, 2026-10-21:
I was also going down this path, but had a different idea -- using a ESP32-S3 as a network bridge to expose the WBH04 pendant on the local network and adding a new pendant type of "WBH04 (Network)" to the Controller app. This has the advantage of working on all devices, but does require additional hardware. Thoughts?

SergeBakharev, 2026-10-21:
I would first investigate adding support directly on the machine firmware. We already have a light USB host there in code, but probably no usb-hid support. 

The ESP32 idea seems like a long communication path which might not be conclusive to a low-latency control experience.