Carvera-Community/Carvera_Controller #286: Feature Request: Enable Auto-Vacuum on CA1 via "Ext. Control"

timbot, 2025-07-20:
Goal:
  Adapt the "Ext. Control" switch to function as a vacuum control for the CA1 model, enabling auto-vacuum functionality.

  Acceptance Criteria:
   1. The feature must be disabled by default.
   2. A user-configurable setting must be available to enable this functionality.
   3. When enabled on a CA1 model, the "Auto Vacuum" switch in the Spindle dropdown should control the "Ext. Control" output.
   4. When enabled, UI elements should refer to this control as "Vacuum" instead of "PowerFan" or "Ext. Control".

  ---

  Development Plan (Proposed)

   1. Add Configuration Setting:
       * Introduce a new boolean setting in `carveracontroller/controller_config.json`.
       * Key: `ca1_ext_control_as_vacuum`
       * Title: Enable Ext. Control as Vacuum (Experimental)
       * Description: For CA1 models, this treats the 'Ext. Control' output as a vacuum and enables the auto-vacuum feature.
       * Default: false

   2. Update UI (`carveracontroller/makera.kv`):
       * Modify the disabled property of the "Auto Vacuum" Switch in the <SpindleDropDown> to be `app.model == 'CA1' and not
         root.ca1_ext_control_as_vacuum`.
       * Update the "PowerFan" label in the <DiagnosePopup> to display "Vacuum" when this feature is enabled: `text: tr._('Vacuum') if
         app.model != 'CA1' or root.ca1_ext_control_as_vacuum else tr._('PowerFan')`.

   3. Update Application Logic (`carveracontroller/main.py`):
       * In the `Makera` class, add a `BooleanProperty` to track the new setting: `ca1_ext_control_as_vacuum = BooleanProperty(False)`.
       * In the `Makera.__init__` method, load the value from the configuration using the appropriate method:
         `self.ca1_ext_control_as_vacuum = Config.getboolean('carvera', 'ca1_ext_control_as_vacuum')`.
       * In the updateStatus method, locate the control_list processing loop. Modify the logic for 'vacuum_mode' to check if the new feature
          is enabled for the CA1 model. If it is, the call should be directed to `self.controller.setExternalControl()` instead of
          `self.controller.setVacuumMode()`.

---

I am willing to implement this feature if it is accepted. Thanks!

timbot, 2025-07-21:
Thank you, @SergeBakharev !

> Please give me a few days to implement this new config management per-machine-model functionality before starting work on this enhancement.

Will do!

> The Diagnostic screen needs to be modified differently than described.

Thanks for the clarifications and additional direction!

timbot, 2025-08-05:
I'm working through this... it still seems like I have to have some device-specific code to load the setting into an instance variable, and there are some timing dependencies.

I'll hopefully have something we can take a look at in the next day or two.

timbot, 2025-08-07:
Here's where I'm at: https://github.com/Carvera-Community/Carvera_Controller/commit/c114d04bb975547bde43005f033d5d02203bc254

The setting loads, and the menus are updated, and we set the external control value when we would have otherwise set the vacuum mode.

What doesn't seem to be working is that activating the setting turns on the vacuum immediately, and it just stays on forever (after the job has completed).

This makes sense after looking at the Makera docs (https://wiki.makera.com/en/supported-codes), the auto-vacuum functionality is actually handled by the firmware, and `M331` just enables it:

```
  M331 Turn on the auto vacuum mode, if on, the vacuum will be turned on automatically when the spindle is running and turned off when the spindle is not running
```

It seems like the best we could do would be to manually interject the external control m-codes (`M851` / `M852`) before/after the job; alternatively, I could look into adding it at the firmware level.

@SergeBakharev WDYT?

I'd also appreciate any feedback on how the setting/config handling is implemented so far.

SergeBakharev, 2025-08-07:
I've opened a draft PR to make it easier to see your changes and comment on the code here: https://github.com/Carvera-Community/Carvera_Controller/pull/320