Carvera-Community/Carvera_Controller #294: Android app often hangs at black screen on launch

sheimend, 2025-07-23:
I am running the carveracontroller-community-25.07.21.apk on a Google Pixel 8. Often when I try to launch the app I get a black screen, after the splash screen, that never advances to the GUI. Closing and restarting sometimes works, but often I can close and relaunch a dozen times in a row before I'll see the actual UI. 

Here's a link to a screen recording of the issue:
https://photos.app.goo.gl/rZjX8sfcUhdtJ5v59

SergeBakharev, 2025-07-24:
Unfortunately this is a bug in the upstream UI framework we use: https://github.com/kivy/python-for-android/issues/2720

We have implemented a work around until they resolve it, but I guess that work around isn't working on your phone. @WARIO2412 do you have any suggestions? Maybe we should schedule a `Window.update_viewport()` a few times during the  first 30 seconds of the app start

SergeBakharev, 2025-07-24:
Increasing the number of forced window renderings resolved the issue on my Pixel device. I've opened https://github.com/Carvera-Community/Carvera_Controller/pull/295

WARIO2412, 2025-07-24:
> I am running the carveracontroller-community-25.07.21.apk on a Google Pixel 8. Often when I try to launch the app I get a black screen, after the splash screen, that never advances to the GUI. [...]

This also happens on my phone. What happens if you don't immediately close the app but go to the homescreen and then switch back to the app again? That's what I do.

SergeBakharev, 2025-07-24:
@sheimend Please try the latest dev build, i've merged code with more aggressive window UI updates for the first 5s of the app start: https://github.com/Carvera-Community/Carvera_Controller/releases/tag/dev

WARIO2412, 2025-07-24:
Fixed for me aswell!

closed 2025-07-24 by sheimend