Carvera-Community/Carvera_Controller #300: Text cut off on small window sizes

jterblanche, 2025-07-27:
When the controller window is made smaller, the text labels in the bar up the top of the screen get cut off.

In the attached photo, the current X position shows 0 (zero), but the actual position is 100.
Similarly with the 3D Probe selected, it just shows 'b' when the window is smaller.

There's no ellipsis hinting that the text has been truncated. This could potentially be fatal for a tool or probe under manual operations,
as the operator could be misled as to the current position of the machine.

[image]

SergeBakharev, 2025-07-28:
Thanks @jterblanche  looks like the minimum widget size I set earlier is still too small.

I wasn't able to reproduce the issue on my systems forcing x1, x1.5, or x2 DPI.  Can you please share the details of your monitor and the display options you are using in the OS? MacOS obscures the exact scaling multiplier detail by default but we can work this out if you have the option "Show all resolutions" enabled:

The difference between native resolution for the monitor and the selected resolution is the DPI multiplier used.

[image]

jterblanche, 2025-08-03:
I can now confirm this happens on MacOS, Windows 11 and Android.

I can also confirm that the Override UI density and Scalng Multipliers have no effect on this.

When I resize the window smaller on desktop platforms the text gets cut off.
My Android doesn't have window resizing but the screen is fairly small and it happens there too.

Let me know what you need me from me or how I can help debug this issue.

SergeBakharev, 2025-09-28:
Fixed with #379

closed 2025-09-28 by SergeBakharev