Carvera-Community/Carvera_Controller #307: Can not single axis probe on Z: Missing required parameter Z Distance

sheimend, 2025-07-30:
This worked in the prior build, but now shows the following error despite me entering a distance for the Z parameter. X and Y do not seem to have this issue.

[image]

SergeBakharev, 2025-07-30:
Can you please confirm the firmware version you are using on the machine?

sheimend, 2025-07-30:
I'm running [community_firmware_v1.0.9c.bin](https://github.com/Carvera-Community/Carvera_Community_Firmware/releases/download/1.0.9c/community_firmware_v1.0.9c.bin)

SergeBakharev, 2025-07-30:
We found the bug. The fix is in the latest [dev build](https://github.com/Carvera-Community/Carvera_Controller/releases/tag/dev). Will give it another few days for any other 0.10.0 related bug reports and will release it as 0.10.1

closed 2025-07-30 by SergeBakharev