Carvera-Community/Carvera_Controller #310: Feature Request: Add Google Drive (OneDrive, etc.) to File Picker locations

sheimend, 2025-07-30:
In my scenario, I'm using an Android phone as my dedicated controller. However, I'm generating all my gcode on my PC. 

Every time I want to upload a new file, I have to go to my phone, disconnect from the controller, go back to my PC, upload the file, disconnect, then reconnect on the phone. (Multi-tennancy would be great but I understand that it's technically complex)

I'm already saving my .cnc files to a drive sync'd via Google Drive. But I do not have access to files stored in Google Drive in the file picker in the Carvera Controller.  It would be great if I did - it would save that full round-trip to my PC!

[I understand that I can separately open Google Drive and _download_ the file locally. However, that doesn't work on iPad, for instance; there's no download option, only "make file available offline", but it's hidden in the file system somewhere inaccessible. Plus this is a big extra step.]

The same request applies to other file sync apps like OneDrive, Dropbox, etc. I suppose. Perhaps the fix is to use an OS-level file picker, rather than the file picker UI in the controller (if that's available)?

SergeBakharev, 2025-08-02:
It looks like Plyer’s [filechooser](https://plyer.readthedocs.io/en/latest/api.html#plyer.facades.FileChooser) is going to be the best way to provide platform specific file selection capability.This should work both on Android and iOS. 

Doing so **should** allow the use of the Google Drive file picker, and other Storage Access Framework Content providers.

This might even allow us to avoid requesting file access permissions to the entire file system. Not sure how this should look in the context of the file browser, maybe the entire "local" side of the explorer is replaced with a single button to open the platform specific file picker, and details of the selected file are shown.

sheimend, 2025-08-03:
Awesome. I think replacing the local side with the platform file picker makes a lot of sense. Thanks so much for looking into this one!

SergeBakharev, 2026-10-14:
As per the comments in #541 I don't think it's practical for us to support such a feature. From what I've seen these cloud storage solutions integrate differently on every OS, and likely slightly differently for each cloud storage vendor. This means we would have to maintain a lot of unique code that might be brittle if the vendor updates thier systems.

Sorry, I'm marking this as a won't fix. It's not realistic for us to do this.

closed 2026-10-14 by SergeBakharev