Carvera-Community/Carvera_Controller #346: Pendant is not detected on macOS if using release dmg

Frostman, 2025-08-28:
Hello,

I've been trying to make WHB04 pendant today on macOS 15.6 (24G84) at MBP M3 using the v0.10.1 release .dmg and it wasn't showing up in the pendant dropdown. If running directly from the source code (from master) using `poetry run python -m carveracontroller` it works just fine.

SergeBakharev, 2025-08-28:
Interesting thanks for the report. Can comment if your system has the hidapi libraries installed separately (eg `brew install hidapi`)?

If not, can you please install them and see if this resolves with the issue with the .dmg packaged application?

SergeBakharev, 2025-09-28:
@acj You are Legend! I was actually in the process of writing a script to download homebrew packages without needing the full client, but I actually prefer to commit the small library files to the repo. This way the build isn't dependent on an environment that has homebrew, and we also have more control over the version of binary that is included.

Do you mind revising #380 commiting the x86_64 and ARM64 versions of libhidapi.dylib to the repo?