Carvera-Community/Carvera_Controller #357: Apple Silicon build of 2.0.0-RC1 hangs when uploading files to machine

byronellis, 2025-09-02:
The Apple Silicon build hangs on trying to upload files to a Carvera Air on the M4 Mac Mini. The Intel build however appears to work fine (M4 MBP).

SergeBakharev, 2025-09-03:
Hi @byronellis ,  thank you for the report. Can you please provide debug level logs from the Apple Silicon system with the issue? See our [documentation on logging](https://carvera-community.gitbook.io/docs/controller/features/logging) for how to enable that.

SergeBakharev, 2025-09-19:
@byronellis were you able to collect logs?

byronellis, 2025-09-19:
Sorry for the delay! Yes, I think I found the logs from when it was happening... They're both pretty short because the app hung. It looks like maybe its more of a machine issue than Apple Silicon, but odd that the Intel version doesn't hang.

[kivy_25-09-02_0.txt](https://github.com/user-attachments/files/22418228/kivy_25-09-02_0.txt)

[kivy_25-09-02_1.txt](https://github.com/user-attachments/files/22418230/kivy_25-09-02_1.txt)

SergeBakharev, 2025-09-24:
Hi @byronellis I'm going to close this issue since I'm pretty sure it has already been resolved. Please reopen if the issue is unresolved in the latest Dev build

closed 2025-09-24 by SergeBakharev