Carvera-Community/Carvera_Controller #364: Reconnect feature assumes wifi connection

SergeBakharev, 2025-09-25:
Reconnect feature assumes wifi connection and will keep retrying the previously used network address until out of timeouts

acj, 2025-09-27:
I have a workaround for this but wanted to ask a few questions before I spend time testing it.

I think the issue is that there are at least two code paths where we can attempt to reconnect when we're on USB. There's a secondary bug where the reconnect timer and dialog box are started, and then we also show the "Reconnection failed. Please connect manually." dialog on top of the countdown dialog. I think the second dialog box is a consequence of the code in the first one -- we try to [cancel reconnecting](https://github.com/Carvera-Community/Carvera_Controller/blob/aa72fd3027ad29ff1ba746a0083156e5357a516a/carveracontroller/main.py#L2893) because we're on USB, which calls the cancel callback (we pass `on_reconnect_failed` for this), which shows the reconnection failed dialog that. Meanwhile, a reconnect timer is still running, so the dialogs keep popping up. Tracing this out was tricky, so it's possible that I misunderstood something, but I think that's the gist of it.

My main question is: do we currently expect auto-reconnect to work on USB? [These lines](https://github.com/Carvera-Community/Carvera_Controller/blob/aa72fd3027ad29ff1ba746a0083156e5357a516a/carveracontroller/main.py#L2892-L2893) make me think that we don't, or that it's somehow handled elsewhere. It doesn't look like the prior USB device name is currently stored, unlike the last-used wifi network. That seems straightforward to add, though, or if there's only a single USB device available then we could try connecting to it. A related issue is that it would be nice for the Reconnect button on the "Connection to machine lost" screen to work with USB instead of being an instant-fail action. These issues seem fixable but may not be trivial.

If we don't want it to auto-reconnect on USB just yet, I think a workaround is to add guards to the code paths I mentioned to check for a wifi connection. I opened #374 for that.

The ideal path, I think, is to make the reconnect feature work with USB as a first-class thing, but I understand if that's too large of a change for an RC and it would be better to wait until after 2.0.0. I'm happy to go either way with it.

@SergeBakharev WDYT?

acj, 2025-09-27:
Sounds good. I'll test it carefully tonight/tomorrow and then mark it as ready.

SergeBakharev, 2025-09-28:
Thanks @acj I've merged #374 into 2.0.0-RC2, let's tackle reconnecting on serial as a 2.1.x enhancement. I think it's a little bit of a different scenario compared to the network connection. 

I don't think the machine resetting actually drops the serial connection as there is a usb-serial chip in between, so it might be less of a "reconnect" and more of "machine stopped responding" situation. It might be better to just increase the machine heartbeat timeout when on serial.